makerbase-mks/MKS-SBASE #98: MKS-SBase v1.3 deprecated and dangerus?

DivingDuck, 2019-03-04:
I just recognize the latest firmware update from Smoothiware tells MKS-Sbase users that their boards are not longer supported.

https://github.com/Smoothieware/Smoothieware/pull/1373/commits/ab1cf6d7ff4bbf3de1f6db8f796ec23d635f64e4

"This is not a sanctioned board and may be unreliable and even dangerous. This MCU is deprecated, and cannot guarantee proper function"

So, who will support the users in the future with actual, save and not outdated firmware releases?

timh0r, 2019-03-09:
switch to marlin 2.x and drop hippie procrastinating selfish community ? it's fully supported there and working fine

HugoW1975, 2019-03-10:
https://www.youtube.com/watch?v=gMPdsBumWCQ&t=152s

And toss the TFT in the bin, hook up a normal LCD (like the Reprap discount full graphics one). I have fought the MKS junk for about a year, now switched to Marlin and the LCD and my prints are fabulous. No more unpredicted stops, no more SD trouble, just good printing.

Hugo

timh0r, 2019-03-10:
https://github.com/timh0r/Marlin/tree/mySGen
this is my profile for sgen/sbase with marlin 2 and tmc2208

DivingDuck, 2019-03-17:
Thanks for your feedback @HugoW1975 and @timh0r.

Well, this wasn't the answer I was looking for. I had never bigger problems with prints or their TFT modules so fare. I'm mostly able to build and troubleshoot 3d printer myself.

I build a couple printers using MKS sbase boards with Smoothieware and mostly like this combination but I'm missing the response of @makerbase-mks here in terms of not answering serious and nicely formulated questions.

It seems to be time to look for other solutions for ongoing projects. What a shame.

timh0r, 2019-03-17:
i doubt they will give you any answer. Basically sbase and sgen are rip off of the smoothie team hard work, except it's opensource hard work and they don't like it at all so they are making false asumptions

Filcanna, 2019-09-23:
To avoid any prob...just stay with older versions of SW firmware (or compile it by yourself). Mine K280 delta works just fine with one of the latest builds, also attached to a Raspberry 3B+/Octoprint.

DivingDuck, 2019-10-28:
Sorry for the late answer. Thanks for your suggestion.

quartapound, 2020-04-22:
> I just recognize the latest firmware update from Smoothiware tells MKS-Sbase users that their boards are not longer supported.
> 
> [Smoothieware/Smoothieware@ab1cf6d](https://github.com/Smoothieware/Smoothieware/commit/ab1cf6d7ff4bbf3de1f6db8f796ec23d635f64e4)
> 
> "This is not a sanctioned board and may be unreliable and even dangerous. This MCU is deprecated, and cannot guarantee proper function"

Check out this link.... https://reprap.org/forum/read.php?418,850453,850459
it appears that they forgot that their original smoothie boards from the kickstarter were also based on LPC1768, not LPC1769, when trying to 'weed out' the clones.... which is why gives you this error.  according to the above link you can safely ignore this, or they'd be effectively eliminating a whole bunch of legitimate users

DivingDuck, 2020-04-23:
I know this fact too. Fortunately I'm able to build the firmware for my own and can also read and/or modify the code  :)

The story behind this question was (and is), that Makerbase should at last take his responsibility and provide their customers with an actual firmware. It is a shame that they leave their (or potential) customer alone with their problems.
I have 5 Sbase as spare parts for some of my customers still here and replaced most of the other boards. In the end (two years ago) I switch to an other supplier as I don't trust both, Makerbase and Smoothieware. I don't want to leave my customers in between the behavior where one never answers questions and the other told them how stupid they are (user or customer ). And what should I say, the support of the new supplier is perfect in all maters.
I still run 3 printer, a small lathe and cnc with one original smoothie and the rest with Sbase boards for my own devices, but I never would give away or sell them with this boards.

makerbase-mks, 2020-06-06:
I suggest that the SBASE motherboard directly uses the marlin2.0 firmware, the available functions will be more, and the compatibility is stronger.

closed 2020-06-06 by makerbase-mks

DivingDuck, 2020-06-06:
Beside to what I said already, I still like Smoothieware much more than Marlin. I am not with you that Marlin have more functions and more compatible as Smoothieware. But sure, in comparison with an obsolete more than 4 years old firmware everything can be told "better". Not to forgot that I (and a lot of people too) bought once your hardware because of Smoothieware. 
Update the firmware and spend a little bit love to your customers (and the development team of Smoothieware). We, your former customers, don't ask for much - only a little bit continuous help here, an actual firmware with a working configuration file (checked from you with your hardware) there, is all they usually want.

It could have been so easy, low hanging fruits all the way ...