makerbase-mks/MKS-SBASE #100: Can't figure out which file is missing

marie-dk, 2019-09-09:
A friend of mine downloaded Marlin from your github page, and followed the guide [Installing Marlin (Sublime)](http://marlinfw.org/docs/basics/install_platformio_sublime.html) and it works well to the point where he is supposed to [run auto-build](http://marlinfw.org/docs/basics/install_platformio_sublime.html#run-auto-build).

The error indicates that a file is missing, but we can't figure out which file. We opened the files in the stacktrace and investigated the line numbers stated in the output, but didn't find any filenames in those lines. How can we determine which file is missing?

The output is:

```
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "buildroot/share/atom/auto_build.py", line 1025, in <lambda>
    self.secondary_thread = threading.Thread(target = lambda q, arg1: q.put(run_PIO(arg1)), args=(que, ''))
  File "buildroot/share/atom/auto_build.py", line 872, in run_PIO
    pio_subprocess = subprocess.Popen(['platformio', 'test', 'upload', '-e', target_env], stdout=subprocess.PIPE, stderr=subprocess.STDOUT)
  File "/usr/lib/python2.7/subprocess.py", line 394, in __init__
    errread, errwrite)
  File "/usr/lib/python2.7/subprocess.py", line 1047, in _execute_child
    raise child_exception
OSError: [Errno 2] No such file or directory

pwd:  /home/[username]/Downloads/MKS-SBASE/MKS-SBASE-master/Firmware/Marlin-bugfix-2.0.x
build_type:   test

Working

python version 2.7.15
build_type:   test
starting platformio
[Finished in 3.6s] 

```

3Dangel, 2019-09-09:
Tell your friend to watch YouTube for “mks sbase Marlin installation”

marie-dk, 2019-09-10:
Do you have any idea what we are missing? If so, please give us a hint. We've searched this error using different terms and no one seems to have experienced this. That is why we made this post as a last resort.

marie-dk, 2019-09-10:
We still don't know what was missing, but he finally had success using Atom instead of Sublime, inspired by [this YouTube video](https://youtu.be/H-c8UTg-EMU). Thanks anyway :-)

closed 2019-09-10 by marie-dk