makerbase-mks/MKS-SBASE #101: MKS sbase with MKS mini 12864 LCD

hussainsail2002, 2019-11-08:
I am unable to get the MKS sbase working with MKS mini 12864 LCD.

I assumed that since both were MKS make it would be compatible. I realized now that the pins are not defined for this LCD.

Could some one help me out with this.

Regards,
Hussain

makerbase-mks, 2019-11-09:
Hi, this you need change it of MKS SBASE（or MKS SGEN） board's EXP1 interface
1. Connect EXP1's pin5 and pin7、pin4 and pin6，As shown below
MKS SBASE(or MKS SGEN) pin map:
[image]
MKS MINI12864 pin map:
[image]
Welding effect picture:
[image]
2. Pin define
Use Marlin fireware
Marlin -> src-> pins -> lpc1768 -> pins_MKS_SBASE.h add pin define
#define DOGLCD_CS      P0_16
#define DOGLCD_A0      P0_15
#define DOGLCD_SCK     P0_07
#define DOGLCD_MOSI    P0_09
[image]
Use smoothieware
[image]
3.Test effect chart
Marlin:
[image]
[image]
Smoothieware:
[image]
[image]

hussainsail2002, 2019-11-11:
@makerbase-mks 

Thanks for the reply,

I have another question. 
In the configuration file I need to activate minipanel right and not mks_mini_12864 ?

hussainsail2002, 2019-11-11:
@makerbase-mks 

My LCD stays on only for few seconds and then shuts off

Any reason for this

Hussain

makerbase-mks, 2019-11-11:
No, you only need enable MKS_MINI12864 .
Can you give me your email address? I will send you my Marlin firmware to you.

hussainsail2002, 2019-11-11:
my email id is hussain@3ding.in

hussainsail2002, 2019-11-11:
I got it working now thanks ! 

I activated mks_mini_12864 instead of minipanel

closed 2019-11-11 by hussainsail2002

makerbase-mks, 2019-11-12:
OK

davigre, 2020-08-09:
Trying to make it work on Smothieware...

I have MKS SBASE L v1 with original MKS Smothieware version: [https://github.com/makerbase-mks/MKS-SBASE/blob/master/English/firmware/firmware.bin](https://github.com/makerbase-mks/MKS-SBASE/blob/master/English/firmware/firmware.bin) (is the only one that has worked for me)

I do not solder pin5 and pin7 and pin4 and pin6, instead I bridge with cable connectors:

**MINI12864      <->       SBASE**
(DOGLCD_AO) <-> (P0.15)
(DOGLCD_CS) <-> (P0.16)
(BTN_NC) <-> (P1.30)
(BEEPER) <-> (P1.31)
(GND) <-> (GND)
(VCC) <-> (VCC)

With the following `config.txt`parameters:
```
panel.enable                                 true            # Set to true to enable the panel code
panel.lcd                                   st7565_glcd     #
panel.spi_channel                           1               # SPI channel to use  ; GLCD EXP1 Pins 3,5 (MOSI, SCLK)
panel.spi_cs_pin                            0.16            # SPI chip select     ; GLCD EXP1 Pin 4
panel.spi_frequency                         500000          # SPI port frequency - some panel need it explicitly set
panel.contrast                              0               # Contrast value for panels that support it ( viki2, mini_viki2 and st7565_glcd )
panel.encoder_resolution                    4               # the number of pulses the encoder emits per detent/click
panel.reverse                               true            # If set to true, reverse the screen.
panel.busy_pin                              nc              # If using the universal_adapter, this pin can be connected to the adapter to ask it if it is busy or not.
panel.a0_pin                                0.15            # (spi A0) If using a viki or SSD1306 this is needed to drive the C/D pin on the display
panel.rst_pin                               nc              # If using an SSD1306 this pin is sometimes required and connects to the reset pin on the display
panel.menu_offset		                    0               #

panel.encoder_a_pin                         3.25!^          # Encoder pin         ; GLCD EXP2 Pin 3
panel.encoder_b_pin                         3.26!^          # Encoder pin         ; GLCD EXP2 Pin 5
panel.click_button_pin                      1.30!^          # Click button        ; GLCD EXP1 Pin 2
panel.buzz_pin                              1.31            # Pin for buzzer      ; GLCD EXP1 Pin 1
#panel.back_button_pin                      2.11!^          # Back button         ; GLCD EXP2 Pin 8

panel.external_sd                           false           # set to true if there is an extrernal sdcard on the panel
panel.external_sd.spi_channel               1               # set spi channel the sdcard is on
panel.external_sd.spi_cs_pin                0.28            # set spi chip select for the sdcard (or any spare pin)
panel.external_sd.sdcd_pin                  0.27!^          # sd detect signal (set to nc if no sdcard detect) (or any spare pin)

panel.menu_offset                            1              # Some panels will need 1 here

panel.alpha_jog_feedrate                     6000           # X jogging feedrate in mm/min
panel.beta_jog_feedrate                      6000           # Y jogging feedrate in mm/min
panel.gamma_jog_feedrate                     200            # Z jogging feedrate in mm/min

panel.hotend_temperature                     200            # Temp to set hotend when preheat is selected
panel.bed_temperature                        70             # Temp to set bed when preheat is selected
```
I notice 'panel.menu_offset' is repeated. Which is the correct value?

I also notice that this parameters are different from here: [https://github.com/makerbase-mks/SGEN_L/blob/master/Firmware/Smoothieware/MKS%20MINI12864%20and%20MKS%20LCD12864A/config.txt](https://github.com/makerbase-mks/SGEN_L/blob/master/Firmware/Smoothieware/MKS%20MINI12864%20and%20MKS%20LCD12864A/config.txt)

The Display turns On but nothing appears. Beeper sounds as soon as I turn on the board or when I push the encoder button.

What more do I need to do?

thiagoedson, 2021-09-24:
MKS_ROBIN_E3D_V1_1  ?