makerbase-mks/MKS-SBASE #102: Mks Robin V2.1 TFT 1.0 Marlin bugfix 2.0 no screen

3Dangel, 2019-11-11:
Would it be possible to get some help with the old Robin’s with TFT 1.0?
Your repository with the working TFT codes is good. But the rest of features are missing. 
Trying to use for Mixing features and the codes in your GitHub are way too old and not have the more complete features to and for mixing extruder with one nozzle. 

Would you be able to help?

Thank you in advance.

makerbase-mks, 2019-11-11:
At present, Marlin does not have a good way to solve the TFT UI display image... full function, if you want to have UI function, you can only use the firmware that comes with Robin. This we will continue to pay attention.

3Dangel, 2019-11-11:
Thank you for your reply. 
Then would it be possible to have the pin out for the board?
At your GitHub repository, I can only find schematic first V2.3 & V2.4 board. 

Thanks you.

makerbase-mks, 2019-11-15:
I will sort out the V2.2 or earlier version and post it to Github.

3Dangel, 2020-03-08:
closing the case.
Main Marlin branch has updated the codes for the ILI 9238 initialization of the TFT screen.
SO, now it is working

closed 2020-03-08 by 3Dangel