makerbase-mks/MKS-SBASE #103: Issues on X surface

aceranic, 2019-11-21:
Hi,

I have issues on x surface. Both sides.
Till now checked mechanics, replaced all bearings, swapped motor, changed current, changed acceleration. Nothing helps.
Z axis is on 12mm rods and separated from screws so no way it is causing wobbling

Any ideas?

Thanks,

aceranic, 2019-11-21:
[image]

3Dangel, 2019-11-21:
May be speed to high.
And it might be your Y axis is you are certain that it is not the Z

aceranic, 2019-11-22:
I reduced speed to 15mm/s and still same. On Y changed bearings and motor. Still same.
Z is separated from screws and is on 12mm linear rails.
I even swapped extruder with different one. Still same.
Driver issue?