makerbase-mks/MKS-SBASE #106: Can't compile with PlatformIO on core.visualstudio

enry68, 2020-03-21:
Hi to all. I've just installed code.visualstudio environment with PlatformIO IDE crosscompiler. I ca'nt compile this version of Marlin because during the compilation i receive this error:

.piolibdeps\LPC1768\TMCStepper_ID5513\src\source\TMC_platforms.h:22:22: error: 'LPC176x' has not been declared

I don't understand. I 've tryied to reinstall the environment but the result is the same. Can kindly someone help me ?

Many thanks.

IN THE FOLLOW THE ENTIRE COMPILE ERROR LIST:

Compiling .pioenvs\LPC1768\lib65d\TMCStepper_ID5513\source\TMC2209Stepper.cpp.o
In file included from .piolibdeps\LPC1768\TMCStepper_ID5513\src\source\SW_SPI.h:4,
                 from .piolibdeps\LPC1768\TMCStepper_ID5513\src\source\SW_SPI.cpp:1:
.piolibdeps\LPC1768\TMCStepper_ID5513\src\source\SW_SPI.cpp: In member function 'void SW_SPIClass::init()':
.piolibdeps\LPC1768\TMCStepper_ID5513\src\source\TMC_platforms.h:22:22: error: 'LPC176x' has not been declared
   22 |   #define writeSCK_H LPC176x::delay_ns(240); LPC176x::gpio_set(sck_pin)
      |                      ^~~~~~~
.piolibdeps\LPC1768\TMCStepper_ID5513\src\source\SW_SPI.cpp:21:3: note: in expansion of macro 'writeSCK_H'
   21 |   writeSCK_H;
      |   ^~~~~~~~~~
.piolibdeps\LPC1768\TMCStepper_ID5513\src\source\TMC_platforms.h:22:46: error: 'LPC176x' has not been declared
   22 |   #define writeSCK_H LPC176x::delay_ns(240); LPC176x::gpio_set(sck_pin)
      |                                              ^~~~~~~
.piolibdeps\LPC1768\TMCStepper_ID5513\src\source\SW_SPI.cpp:21:3: note: in expansion of macro 'writeSCK_H'
   21 |   writeSCK_H;
      |   ^~~~~~~~~~
.piolibdeps\LPC1768\TMCStepper_ID5513\src\source\SW_SPI.cpp: In member function 'uint8_t SW_SPIClass::transfer(uint8_t)':
.piolibdeps\LPC1768\TMCStepper_ID5513\src\source\TMC_platforms.h:23:22: error: 'LPC176x' has not been declared
   23 |   #define writeSCK_L LPC176x::delay_ns(240); LPC176x::gpio_clear(sck_pin)
      |                      ^~~~~~~
.piolibdeps\LPC1768\TMCStepper_ID5513\src\source\SW_SPI.cpp:26:3: note: in expansion of macro 'writeSCK_L'
   26 |   writeSCK_L;
      |   ^~~~~~~~~~
.piolibdeps\LPC1768\TMCStepper_ID5513\src\source\TMC_platforms.h:23:46: error: 'LPC176x' has not been declared
   23 |   #define writeSCK_L LPC176x::delay_ns(240); LPC176x::gpio_clear(sck_pin)
      |                                              ^~~~~~~
.piolibdeps\LPC1768\TMCStepper_ID5513\src\source\SW_SPI.cpp:26:3: note: in expansion of macro 'writeSCK_L'
   26 |   writeSCK_L;
      |   ^~~~~~~~~~
.piolibdeps\LPC1768\TMCStepper_ID5513\src\source\TMC_platforms.h:20:23: error: 'LPC176x' has not been declared
   20 |   #define writeMOSI_H LPC176x::gpio_set(mosi_pin)
      |                       ^~~~~~~
.piolibdeps\LPC1768\TMCStepper_ID5513\src\source\SW_SPI.cpp:30:28: note: in expansion of macro 'writeMOSI_H'
   30 |     !!(ulVal & (1 << i)) ? writeMOSI_H : writeMOSI_L;
      |                            ^~~~~~~~~~~
compilation terminated due to -fmax-errors=5.
*** [.pioenvs\LPC1768\lib65d\TMCStepper_ID5513\source\SW_SPI.cpp.o] Error 1

enry68, 2020-03-21:
EDIT: I've solved simply substituing the actual TMC library 6.01 with an old one (5.0) in my case and now it works. I hope this can be useful for other users.

Enrico.

makerbase-mks, 2020-04-02:
OK! Generally this kind of error is basically caused by the library .

closed 2020-04-02 by makerbase-mks