makerbase-mks/MKS-SBASE #110: Last Smothieware firmware doesn't work

davigre, 2020-08-09:
I'm using MKS-SBASE with Smothieware version: "Build version: edge-823faf1, Build date: Dec 27 2014 04:05:24, MCU: LPC1768, System Clock: 100MHz" and is working fine as a DELTA Printer driver.

I want to mount last Smothieware version so I mount this version: [https://github.com/Smoothieware/Smoothieware/blob/edge/FirmwareBin/firmware.bin](https://github.com/Smoothieware/Smoothieware/blob/edge/FirmwareBin/firmware.bin)

And I noticed blocking (freezing) problems when sending several GCODE's.

So then I return to the one in this repository: [https://github.com/makerbase-mks/MKS-SBASE/blob/master/English/firmware/firmware.bin](https://github.com/makerbase-mks/MKS-SBASE/blob/master/English/firmware/firmware.bin)

And started to work again.

Reading [issue 92](https://github.com/makerbase-mks/MKS-SBASE/issues/92) it says that I can use any Smothieware firmware, but I can't.

I also found the (MAKERBASE original) Smothieware source code here: [https://github.com/makerbase-mks/MKS-SBASE/blob/master/English/Source/Smoothieware-edge.rar](https://github.com/makerbase-mks/MKS-SBASE/blob/master/English/Source/Smoothieware-edge.rar), but to much work for me.

Please indicate me which is the last Smothieware firmware that I can use.

I don't want to use Marlin, I love 'config.txt' file configuration.

Thanks.

3Dangel, 2020-08-09:
Than purchase the original Smoothie board in order to be compatible.
MKS Sbase is NOT  supported by smoothie community.

DivingDuck, 2020-08-09:
@davigre, had you use your old config file with the latest firmware?

davigre, 2020-08-12:
@DivingDuck , Yes. I don't change it.

@3Dangel , I already start to learn Marlin. it looks too many configuration parameters. But I think is the way to go. Thanks.

DivingDuck, 2020-08-12:
@davigre it's maybe worth to take over your values in a newly downloaded config.txt. I'm running the latest build on an core-xy both with SD card and USB via Pronterface (and Cura too) without any problem.

davigre, 2020-08-15:
@DivingDuck ,

Thank's for encourage me. It finally works. 

The problem was not 'config.txt' file . I discovered that when I send M105 the following message was output:

```
MINTEMP or MAXTEMP triggered on B. Check your temperature sensors! HALT asserted - reset or M999 required
```
Searching, I found this: [http://smoothieware.org/temperaturecontrol#maximum-temperature-detection](http://smoothieware.org/temperaturecontrol#maximum-temperature-detection)

Found that my 'bed' thermistor was wrongly defined as 'Honeywell100K'. I really don't know which is my thermistor type, but looking [http://smoothieware.org/steinharthart](http://smoothieware.org/steinharthart) I measure Resistance and take room temperature putting as my first value in the example that I found in that page:

```
temperature_control.hotend.rt_curve          20.0,116700,150,1360,240,206.5
```

I know is not perfect but it works!!!. I will buy a Digital Infrared Thermometer to increase precision.

Found also when executing 'version' command the following output:

```
Smoothie command shell
version
Build version: upstreamedge-c0d4117, Build date: Jul 31 2020 17:46:39, MCU: LPC1768, System Clock: 100MHz
5 axis
      NOTICE: This MCU is deprecated, and cannot guarantee proper function
```

Thanks.

closed 2020-08-15 by davigre

DivingDuck, 2020-08-18:
Thanks for your feedback. Happy printing.