makerbase-mks/MKS-SBASE #111: MKS SBASE (v1.3) + MKS mini 12864 LCD (v1.0) + SMOOTHIEWARE (version Jul 31, 2020)

davigre, 2020-08-16:
I have a MKS SBASE v1.3 working with Last Smoothieware version + MKS TFT 32 + Octoprint.

I buy MKS SBASE + MKS mini 12864 on Feb of 2018, but was until now that I have time to mount both items in my Delta printer.

The only thing that rest to mount and finish this project is the 'MKS mini 12864 V1.0'.

I already found that is not a 'only connect cables' thing. But I know that can be done because the photos on #101 

I follow #101

and checking [MKS mini 12864 pint outs](https://reprap.org/wiki/File:MKS_MINI12864_LCD_controller_pin_out_-_signal_names.jpg) 

I do not solder the SBASE (need to unmount a lot of things to do that) so what I do was connect with cables to the required pins on **EXP1**:

**MINI12864 <-> SBASE**
(DOGLCD_AO) <-> (P0.15)
(DOGLCD_CS) <-> (P0.16)
(BTN_ENC) <-> (P1.30)
(BEEPER) <-> (P1.31)
(GND) <-> (GND)
(VCC) <-> (VCC)

**EXT2** without modifications, connected with ribbon cable.
[image]

I don't need SD, so I do not enable it in 'config' file.

```
panel.enable                                true            # Set to true to enable the panel code
panel.lcd                                   st7565_glcd     # reprap_discount_glcd, st7565_glcd, ssd1306_oled, viki2, mini_viki2 and universal_adapter
panel.spi_channel                           0               # SPI channel to use  ; GLCD EXP1 Pins 3,5 (MOSI, SCLK)
panel.spi_cs_pin                            0.16            # SPI chip select     ; GLCD EXP1 Pin 4
#panel.spi_frequency                         500000          # SPI port frequency - some panel need it explicitly set
panel.contrast                              9               # Contrast value for panels that support it ( viki2, mini_viki2 and st7565_glcd )
panel.encoder_resolution                    4               # the number of pulses the encoder emits per detent/click
panel.reverse                               true            # If set to true, reverse the screen.
panel.busy_pin                              nc              # If using the universal_adapter, this pin can be connected to the adapter to ask it if it is busy or not.
panel.a0_pin                                0.15            # (spi A0) If using a viki or SSD1306 this is needed to drive the C/D pin on the display
panel.rst_pin                               nc              # If using an SSD1306 this pin is sometimes required and connects to the reset pin on the display
panel.menu_offset		                    0               #
panel.encoder_a_pin                         3.25!^          # Encoder pin         ; GLCD EXP2 Pin 3
panel.encoder_b_pin                         3.26!^          # Encoder pin         ; GLCD EXP2 Pin 5
panel.click_button_pin                      1.30!^          # Click button        ; GLCD EXP1 Pin 2
panel.buzz_pin                              1.31            # Pin for buzzer      ; GLCD EXP1 Pin 1
#panel.back_button_pin                       2.11!^          # Back button         ; GLCD EXP2 Pin 8

panel.external_sd                           false           # set to true if there is an extrernal sdcard on the panel
panel.external_sd.spi_channel               0               # set spi channel the sdcard is on
panel.external_sd.spi_cs_pin                0.28            # set spi chip select for the sdcard (or any spare pin)
panel.external_sd.sdcd_pin                  0.27!^          # sd detect signal (set to nc if no sdcard detect) (or any spare pin)
panel.menu_offset                           0               # Some panels will need 1 here

panel.alpha_jog_feedrate                    4000            # X jogging feedrate in mm/min
panel.beta_jog_feedrate                     4000            # Y jogging feedrate in mm/min
panel.gamma_jog_feedrate                    4000            # Z jogging feedrate in mm/min

panel.hotend_temperature                    200             # Temp to set hotend when preheat is selected
panel.bed_temperature                       60              # Temp to set bed when preheat is selected
```

I already try changing different parameters in my config.

The good news is that MKS MINI 12864 turns On and ENCODER, BEEPER works also. Even the 'STOP' button. I know because I play with the encoder and the SBASE is executing commands. So the only thing that is not working is the display.

@makerbase-mks, can you check please. What I'm doing wrong?

davigre, 2020-08-21:
Finally the key was this [image](http://chibidibidiwah.wdfiles.com/local--files/panel/smoothieboard-LCD_1024.jpg)

**MKS mini 12864 Connection (plus the normal ones. NC = Do not connect ):**
[image]

**And the configuration ('config' file):**
```
panel.enable                                true            # Set to true to enable the panel code
panel.lcd                                   st7565_glcd     # reprap_discount_glcd, st7565_glcd, ssd1306_oled, viki2, mini_viki2 and universal_adapter
panel.spi_channel                           0               # SPI channel to use  ; GLCD EXP1 Pins 3,5 (MOSI, SCLK)
panel.spi_cs_pin                            0.16            # SPI chip select     ; GLCD EXP1 Pin 4
panel.contrast                              0               # Contrast value for panels that support it ( viki2, mini_viki2 and st7565_glcd )
#panel.encoder_resolution                   2               # the number of pulses the encoder emits per detent/click
panel.reverse                               true            # If set to true, reverse the screen.
#panel.busy_pin                              nc              # If using the universal_adapter, this pin can be connected to the adapter to ask it if it is busy or not.
panel.a0_pin                                2.11            # (spi A0) If using a viki or SSD1306 this is needed to drive the C/D pin on the display
#panel.rst_pin                               nc              # If using an SSD1306 this pin is sometimes required and connects to the reset pin on the display
panel.menu_offset		             1               # On some panels, this value must be set to 1. This is a number of lines to offset the menu lines by on screen.
panel.encoder_a_pin                         3.25!^          # Encoder pin         ; GLCD EXP2 Pin 3
panel.encoder_b_pin                         3.26!^          # Encoder pin         ; GLCD EXP2 Pin 5
panel.click_button_pin                      1.30!^          # Click button        ; GLCD EXP1 Pin 2
panel.buzz_pin                              1.31            # Pin for buzzer      ; GLCD EXP1 Pin 1
panel.back_button_pin                       4.28!^          # Back button         ; GLCD EXP2 Pin 8

panel.external_sd                           false           # set to true if there is an extrernal sdcard on the panel
panel.external_sd.spi_channel               1               # set spi channel the sdcard is on
panel.external_sd.spi_cs_pin                0.28            # set spi chip select for the sdcard (or any spare pin)
panel.external_sd.sdcd_pin                  0.27!^          # sd detect signal (set to nc if no sdcard detect) (or any spare pin)
```
Notice that I do not enable 'external_sd', no problem, you can access MicroSD from the board.

[image]

closed 2020-08-21 by davigre

augustenlee, 2021-06-25:
HELLO ,
I found the something, why your P0.15 is different form mine?
[image]
But I think your pin map is right..
Because I found out from the source code:
[image]
it defines the SPI channel by default 2 set in it , I think this is the reason you have to change SPI Channel and re-porting your pins.