makerbase-mks/MKS-SBASE #112: VALUE OF Rsense OF DRIVER DR8825 to calculate Vref  FOR MOTOR STEPPER

bilou24an199, 2021-02-04:
hello i have a question concerning the DR8825 installed on the MKS SBASE v1.3
i have some troubles with 1 motor step on my 3d printer and i want to be sure that the value of voltage and current is ok for the motor
to do that the  formula is  => Vref = Inom x 5 x Rsense
the Rsense is indicated on drivers. Normally, it is easy to find on an external driver but here the drivers are included on the PCB.
have take a look on the MKS SBASE V1.3_SCH.pdf but if I make a mistake, it will not be good for the motors.
can you give me the value of Rsense included on the board for each driver

axis X / Axis Y / Axis Z / E0 and E1

example = for driver axis X, list of resistor (which is the the good value)
Rm7_M2 = 0.1
Rm8_M2 = 0.1
Rm1_M2 = 10k
Rm4_M2 = 1.5k
Rm3_M2 = 10k
Rm2_M2 = 100k

For the moment I think that the value are = Rm7_M2 = 0.1 / Rm8_M2 = 0.1

thanks for your support

BR

mks-viva, 2021-02-06:
Hi! Rense is Rm7 and Rm8 = 0.1Ω

bilou24an199, 2021-02-07:
thanks a lot for your support

BR

closed 2021-02-07 by bilou24an199