makerbase-mks/MKS-SBASE #114: Extruder does not work while printing

qulet, 2021-08-11:
I bought a MKS-Sbase V1.3 board for a 3D printer.
X, Y, Z axis control works fine. There is a problem with the E axis - extruder. It can be properly calibrated with the LCD steps / mm x 100 mm of the filament. Works fine "extrude" and "retract". Everything seems "OK". The problem occurs while printing. The extruder engine on a correctly generated g-code Simplifay3D performs "extrude" and "retract" with the same number of steps / mm oscillates as a result the filament is not extruded ........

It is not a problem of a clogged nozzle or setting the correct height in the "Z" axis 
I checked it on Marlin V2.09.01 and Smoothieware 2020. The flaw is hidden in the hardware. I am asking for help in fixing this problem.

Quas7, 2021-08-11:
Does the same happen using E1 instead of E0?

qulet, 2021-08-15:
[marlin_2.9.0.1.zip](https://github.com/makerbase-mks/MKS-SBASE/files/6988783/marlin_2.9.0.1.zip)
Configuration files and compilation ...

qulet, 2021-08-15:
58 / 5000
Wyniki tłumaczenia
Sorry, the problem is current, I'm sending a test G-code 
[30x30.zip](https://github.com/makerbase-mks/MKS-SBASE/files/6988790/30x30.zip)

qulet, 2021-08-15:
Na RAMPS 1.3  wszystko działa "OK"
[image]
[image]

DivingDuck, 2021-08-16:
If I was you, I would try to diagnostic the problem different. 

Put the controller on your desk, take only one motor and and test movement with it for each axis from x to e1. This should be easy to do if you connect the board to your power supply and your computer via USB. Use Pronterface for interacting. Don't forget to disconnect the controller from power each time before you switch the motor connector to an other axis ( ... and take this serious, you will definitely damage the motor driver). In case a axis do not move you have either - a problem in hardware or with firmware configuration.

In case of smoothieware, don't use the mks version, use the original smoothieware edge version from github in addition a new downloaded configuration file from there as well and set it up new from first to last line following the instructions from the smoothieware documentation. Take care with the pin definitions. Some are different between mks and an original smoothieboard, so you need to change those accordingly.

A comment for extruder (s): There is usually these days a thermal runaway protection that prevent an E-stepper to turn if the heater don't have a decent temperature. Maybe worth to check too. For testing you can disable this function, but don't forget to activate it again after your test as this is an important safety function.

And a last comment/tip for your controller board assembly. Take care you have enough air space between your assembly plate and controller. The heat sinks of the motor drivers do suggest a user that the major heat desperation will happen there. This is wrong, it will happen below to the board and if you don't have a constant air flow underneath the board you will become sooner or later problems with thermal shut dow's of motor driver's during printing.

Good luck!

qulet, 2021-08-16:
Wysłane z aplikacji Poczta dla systemu Windows  Od: DivingDuckWysłano: poniedziałek, 16 sierpnia 2021 10:36Do: makerbase-mks/MKS-SBASEDW: qulet; State changeTemat: Re: [makerbase-mks/MKS-SBASE] Extruder does not work while printing (#114) If I was you, I would try to diagnostic the problem different.Put the controller on your desk, take only one motor and and test movement with it for each axis from x to e1. This should be easy to do if you connect the board to your power supply and your computer via USB. Use Pronterface for interacting. Don't forget to disconnect the controller from power each time before you switch the motor connector to an other axis ( ... and take this serious, you will definitely damage the motor driver). In case a axis do not move you have either - a problem in hardware or with firmware configuration.In case of smoothieware, don't use the mks version, use the original smoothieware edge version from github in addition a new downloaded configuration file from there as well and set it up new from first to last line following the instructions from the smoothieware documentation. Take care with the pin definitions. Some are different between mks and an original smoothieboard, so you need to change those accordingly.A comment for extruder (s): There is usually these days a thermal runaway protection that prevent an E-stepper to turn if the heater don't have a decent temperature. Maybe worth to check too. For testing you can disable this function, but don't forget to activate it again after your test as this is an important safety function.And a last comment/tip for your controller board assembly. Take care you have enough air space between your assembly plate and controller. The heat sinks of the motor drivers do suggest a user that the major heat desperation will happen there. This is wrong, it will happen below to the board and if you don't have a constant air flow underneath the board you will become sooner or later problems with thermal shut dow's of motor driver's during printing.Good luck!

DivingDuck, 2021-08-16:
A retraction is only a negative movement on the E- axes, so if you say the E-axis motor is moving, there can't be a hardware problem of your controller board causing this. Well, at least I can't imagine how this can be a case.
I assume, you check that your hotend have the correct temperature before extruding filament.

Oscillate means the extruder motor try to turn but can't, make instead some ugly noises?

This can happen if the motor current is too low or the filament is sticking somewhere between extruder and hot-end nozzle in your filament path. In the lather you can hear that your motor is clicking and the filament jumps back a little bit. Maybe also if you set up much too high speed and/or acceleration values.

An other thing what can cause all kind of problems is a defect gcode file / memory card. Quick to check.
Maybe also worth to check if the problem occurs too if you disconnect your LCD and run the print directly from Printrun via USB.
Also, you have set up the correct gcode flavor in your slicer for generating your gcode? This shouldn't cause your problem, but who knows, better to check it too...

So, now I'm out of ideas to look at for your problem.

qulet, 2021-12-03:
Zmiana  wartości  Retraction speed 3600 mm/minna 1800 mm/min w w slicer usuneła problem

closed 2021-12-03 by qulet